SPIRIT-org/SPIRIT #13: Chassis: CAD Model, technical Drawings, Manufacturing Formats

MavLeague, 2025-08-10:
Todo:
- [ ] CAD Model
- [ ] technical Drawings for Product Qualifications
- [ ] Tool paths etc. for CNC machining

CAD Software: Onshape
- Public Project
- Browser based
everyone can follow the design process and see where I'm at

boelle, 2025-08-10:
Q: can others also edit or suggest edits ? i plan to invest in the hardware at some point if someone will help me get latest android working on it

my aim was to replace my old samsung with something i can fix myself if things break or wear out

MavLeague, 2025-08-10:
Yes, that's the plan. Onshape offers a version-history and is also made for collaboration with others. So I hope that it has the most benefit for this project. Also I am to be honest not really familiar with the workflow in fusion. I'm not sure if it offers the same benefits Onshape does.

boelle, 2025-08-10:
it does not have versio history, but you can give access to other people so they can work on the same model

a yes it does have some kind of version history in that each time you save the document gets a higher version number and you can revert to a version in the past if you F'up things

MavLeague, 2025-08-10:
Okay, but the workflow isn't like the cad workflow I am used to. Also I am not sure if public Projects are easily possible. So I think it's best to stick to Onshape for now.

But don't worry, suggestions are of course welcome!

MavLeague, 2025-08-13:
```mermaid
flowchart LR
    A[General 3D model<br/>based on PCB] --> B[Derive]

    B --> C[Optimize for 3D printing]
    C --> D[Slice]

    B --> E[Optimize for CNC Milling]
    E --> F[Technical Drawing]
    F --> G[Write CNC machining program]

MavLeague, 2025-08-13:
I've created a little Roadmap to show what my Plan is:

```mermaid
flowchart LR
    A[General 3D model<br/>based on PCB] --> B[Derive]

    B --> C[Optimize for 3D printing]
    C --> D[Slice]

    B --> E[Optimize for CNC Milling]
    E --> F[Technical Drawing]
    E --> G[Write CNC machining program]

MavLeague, 2025-08-16:
> Can anyone contribute to this?
> 

It's a bit more complicated, but my goal is to explain that in the wiki. In a nutshell, we don't use GitHub but Onshape for creating the chassis so we have to approach it differently.

V3lectronics, 2025-08-20:
cm5 has been put to the back for better thermals:

[image]

koishi-514-shutdown, 2025-08-21:
希望外壳的摄像头不要占用背板大量面积，就像以前的手机一样简洁像iPhone6一样，并且可以在桌面上放平。
关于cm5的散热问题，我觉得可以在外壳相应位置提供类似铜或铝垫片来提供被动散热
希望可以有对应的3d打印手机壳模型
建议手机外壳设计的时候考虑电池的拆卸方便，设计一个结构可靠简单的外壳

V3lectronics, 2025-08-22:
button clearance fixed!

MavLeague, 2025-08-26:
There's now a wiki entry according to this issue.

This is a snippet of that:

## Designing Process

### CAD Software

You probably wonder where and how we create the model. We use Onshape as our CAD Program of choice because it makes collaboration really easy.

### View Progress

To see where we're at you can access the [Public Project at Onshape.](https://cad.onshape.com/documents/4e52c2512b67af09635f94b3/w/56ef775ef64c6c9cfce93a1f/e/4e6df7991d5269c6c76f6984)

To view it you **need** an Onshape Account (but the free plan is enough).

### Contribute
[_coming soon..._]

cholupa, 2026-09-28:
I am sure this is already thought of, but what about a small tactile button?

MavLeague, 2026-09-28:
> I am sure this is already thought of, but what about a small tactile button?

It's already been implemented by now, but thanks for the input!

cholupa, 2026-09-28:
No problem and appreciate the response, I was just sifting through the open issues.

This is a really cool project.

Cheers

Sent from [Proton Mail](https://proton.me/mail/home) for Android.