SPIRIT-org/SPIRIT #18: GSM/GPS section (EC25 GSM/GPS, SIM holder)

V3lectronics, 2025-08-10:
We need someone to deal with the GSM/GPS schematic in KiCad

Main chip: https://www.digikey.pl/en/products/detail/quectel/EC25VFA-512-STD/13278151

[image]

BigBOSSJake, 2025-08-13:
Witch form factor do you like , I have a few Global options,and you want a phone network or just GPS?

BigBOSSJake, 2025-08-14:
The one that you have found is not vary global or easy to remove , a M.2 one is better for a lot of resons.

boelle, 2025-08-14:
hmm... would it not be the smartest to have wifi and gps seperate so you can update only one if say we get wifi 8 but that we still only have LTE 5G, no reason to update the GSM part then

V3lectronics, 2025-08-15:
Are we settling on eg25 as an upgrade to ec25? Maybe there are other suggestions?

BigBOSSJake, 2025-09-17:
Are we done ?