SPIRIT-org/SPIRIT #20: POWER section (IN PROGRESS)

V3lectronics, 2025-08-10:
Help needed with implementing the power section schematic.

[image]

DansDesigns, 2025-08-29:
what are your main power requirements? are you running 5v into the CM? do you need 3v3 and 1v8 anywhere on the board?

I decided to use an off-the-shelf power converter that takes a native 1S lipo and converts it to 5v 3A, from there i am using 3v3 LDOs for local power where needed and to the CM for main 5v power...

V3lectronics, 2025-09-02:
Evaluation module from TI with the BQ25798:
https://www.ti.com/lit/ug/sluucb5e/sluucb5e.pdf?ts=1756794755662

Very useful, includes schematics and payout!!!

MrPouette, 2025-09-17:
We where looking to blink the R1202A regulator to modulate the Backlight brightness in the Youtube Live. But that would modulate the whole rail. 
If it's only for the lcd, it seem to be a good solution to use less part but that would require to change the R1202A for a R1202D. Otherwise, we will modulate the output by the Output Capacitor Discharge Fet trough the CE pin(I think please take another look to the block diagram in datasheet).

Are we trying to conceive a 22V power rail for multiple part or only for the screen(RK055AHD091) ?

V3lectronics, 2025-09-19:
> Are we trying to conceive a 22V power rail for multiple part or only for the screen(RK055AHD091) ?
The 22V rail is only for the screen

as for the A/D version I think you're right but I'm not 100% sure.

V3lectronics, 2025-10-02:
Implemented, double check needed

V3lectronics, 2026-01-31:
Tps650250 will soon be replaced due to driver support

closed 2026-03-03 by V3lectronics