SPIRIT-org/SPIRIT #31: Touchscreen

V3lectronics, 2025-08-13:
Concerning the discussion about upgrading from https://www.digikey.com/en/products/detail/nxp-usa-inc/RK055HDMIPI4MA0/16274141

to

https://focuslcds.com/product/e55rb-i-mw400-c/

### Low stock

I sent an email to FocusLCDs asking if they plan to increase their their stock (5 units at the time of writing). Here is their response: 

> Hey Jan,
>
>Most of our customers are in the medical, industrial, automotive and defense industries and long product life is very important.  While the lifecycle is different for each display depending on the components used, for most items listed online, Focus LCDs expects at least 3 – 5 years of availability.  In many cases, displays are available for longer than this, but it is difficult to forecast.  In the event a display does go obsolete, we work to provide samples of replacements to purchase.  These are often drop-in replacements and require no changes to the firmware of the product.
>
>E55RB-I-MW400-C is an active product and can be purchased through [our website](https://focuslcds.com/product/e55rb-i-mw400-c/) with no MOQ

I.e. we don't know if the stock will be increased and 5 units is just too little to bet on. The 5 available are probably just samples for companies that plan to place big orders.

### Cost

The prices are similar, difficult to judge because idk if their website price includes VAT. Overall the FocusLCDs is slightly more expensive. 

### Logistics

The bigger issue is with logistics: all other components are available from Digikey so they can be sent in one package (maybe except the battery which cannot be sent by air).

Additionally, Digikey provides free shipment for large orders (components for a single SPIRIT are expensive enough to get free shipping from Digikey).

Tldr: it makes sense cost-wise to order all components from one distributor if possible!
 
### Decision

For now we stick to https://www.digikey.com/en/products/detail/nxp-usa-inc/RK055HDMIPI4MA0/16274141

(If SPIRIT were a mass produced device I would take the other option)

V3lectronics, 2025-08-14:
https://www.waveshare.com/5dp-caplcd.htm 

has been proposed as an alternative to 

https://www.digikey.pl/en/products/detail/nxp-usa-inc/RK055HDMIPI4MA0/16274141

### 1 Too much IO

All of the IO on the carrier PCB of this screen takes too much space compared to just having one MIPI interface. 

### 2 Not available on Digikey

For easier logistics and free shipping it is much more efficient to order all components from one distributor.

### 3 Lower specs

The resolution AND screen size are smaller

Engineer162, 2025-08-16:
I found a screen on AliExpress (the goat) that seem to be worth looking at as it has a higher resolution, is cheaper, and is made by a company with some decent reputation.

47 EURO 1080P AMOLED touch screen: https://www.aliexpress.com/item/1005004285318699.html?spm=a2g0o.store_pc_home.promoteRecommendProducts_15459622.1005004285318699

Yes, I know it's from AliExpress, but the seller Wisecoco has a huge selection to choose from and for a good price. Also, we will have to get the pinout for the 40 pin FPC connector, although I'm sure the seller is willing to provide that.

Here are also links to the 2 known Wisecoco stores.

First store: https://wisecocoscreen.aliexpress.com/store/311228?spm=a2g0o.detail.0.0.5c3birfXirfXQH

Second store: https://www.aliexpress.com/store/4682114?spm=a2g0o.detail.0.0.e1019thX9thXq0

LetsPlaySimi, 2025-08-16:
I've found what seems to be the same screen but higher resolution for cheaper.

https://de.aliexpress.com/item/1005006270340420.html

I know it looks sus but the seller has good reviews, many sold items, and even if it doesn't work, AliExpress offers free returns and refunds your payment if it's broken. Also AliExpress pricing is just weird.

Engineer162, 2025-08-16:
I looked at that screen before, the only problem is that I does not have touch. Otherwise I would have linked to that one instead of the Amoled one.

V3lectronics, 2025-08-19:
> is cheaper

It's cheaper because it doesn't have the driver included !!! (almost got me XD)

also keep in mind you need to pay for shipping and the price increases with each unit, very bad for a (perhaps-one-day-massproducable device). Digikey ships for free if we order all parts from them.

@Engineer162

Engineer162, 2025-08-19:
Ah that’s just the screen mipi to hdmi converter and controller, the current screen you are using im fairly certain already uses mipi. So this screen should already function out of the box with the cm5 as long as it is correctly pin mapped, as it uses the mipi standard interface with touch over IC2 (I think). I will most likely purchase this screen and test it with my Pi 5 to make sure everything works.

@V3lectronics

Engineer162, 2025-08-20:
We don’t need that board nor that white cable, as it’s converting from the screens mipi to hdmi. Because the current design of the phone already uses mipi, we don’t need the board as you are using the same protocol as this 1080p display. Which means in theory all that is needed to adapt this screen from the old one is to change the connector type on the cm5 carrier.

Nevertheless I will be ordering the display (without the controller board) and testing it. Expect this to take at least a month, as shipping takes quite a while to Denmark.

Engineer162, 2025-08-22:
Here is the datasheet for that display

[TOP055FHDT00CTP01-5.5 OLED.pdf](https://github.com/user-attachments/files/21944957/TOP055FHDT00CTP01-5.5.OLED.pdf)

Secret-chest, 2025-09-03:
With the Waveshare I saw a problem that they have some standoffs for the large Pi that might be bad for this design.