SPIRIT-org/SPIRIT #32: 3D Model for Touchscreen

MavLeague, 2025-08-13:
For designing the chassis, I need a 3D Model of the Touchscreen we'll use.

Wich Touchscreen we'll use will be decided in issue #31 so we propably will need to wait for that decition first.

Requirements:
- STEP File format
- simple colorization
- unit: mm

c3am, 2025-08-13:
Hi, I would like to make the 3d model of the screen.
Why is it necessary though to have two versions? Don't we just need the one with the ribbon cable tucked under the screen?

c3am, 2025-08-13:
Also, how do I upload the file? github tells me that .step and .FCStd files are not supported in this messaging thing. Should I put the file into `main/mechanical/lib` for example and make a pull request?

boelle, 2025-08-14:
> Should I put the file into `main/mechanical/lib` for example and make a pull request?

EDA-kicad/component-libs/SPIRIT-3D-models

MavLeague, 2025-08-14:
> Why is it necessary though to have two versions? Don't we just need the one with the ribbon cable tucked under the screen?

Because the first version is the default how it's purchased and the second one is the one how it's mounted

MavLeague, 2025-08-14:
@c3am, could you contribute today so that I can start integrating the screen into the chassis? Alternatively, you could send me the model on Discord and add the pull request later.

closed 2025-08-14 by MavLeague