SPIRIT-org/SPIRIT #33: Sugestion for SDR/flipper0 like recorder

vertigofilip, 2025-08-13:
I remember different project for cyber deck on YouTube, and there was a small SDR used in there, that could potentially fit in this device.  The SDR is NESDR Nano 3 OTG. Potential problems might be with connecting (SDR is USB device), with cooling (it has external radiator on its enclose, or with range (it has external antenna), but I think it will work relatively easy, and it has good range, and range of frequencies it can scan.

vertigofilip, 2025-08-14:
In the parts list there was mentioned flipper0 like recorder, so I thought it was supposed to be something like this.

V3lectronics, 2025-08-20:
so I think it's best to make this a separate external addon