SPIRIT-org/SPIRIT #36: consider Postmarket OS for OS

6mat, 2025-08-13:
You mentioned exploring developing an os based on Debian. Postmarket OS is alpine Linux based.

easton57, 2025-08-13:
Had this thought after seeing the github. Looks like there's already a wiki page available for it. As long as the drivers for the screen and other peripherals are mainlined then it should be a fairly simple port. https://wiki.postmarketos.org/wiki/Raspberry_Pi_5_(raspberry-pi5)

BigBOSSJake, 2025-08-13:
I think, that any os that works on raspbarry py 5(or any cm) will work(like even Windows).

easton57, 2025-08-13:
Sweet. I can make a PR to add it to the readme.

boelle, 2025-08-14:
you can put any OS on it, its just a matter of what work you have to put in it for it to work

6mat, 2025-08-15:
Thank you, awesome.
I like the feel of the OS. Feels like Linux