SPIRIT-org/SPIRIT #47: Windows full suport and more security.

BigBOSSJake, 2025-08-15:
I think that TPM would be a good idea for it increases the securty of the os.

Adtsd24, 2025-08-15:
Dude, this is not a PC, this is supposed to be a smartphone.

berkerogluu, 2025-08-15:
This is not a PC ? For Linux, enabling kernel lockdown mode is sufficient.

Axel84687, 2025-08-16:
I feel like we wanting to include so much things on the software, why it can just be android simple

MarcSN311, 2025-08-24:
That video is a mix of legit concerns, speculative stuff presented as fact, and some full-on tinfoil hat territory. It also doesn’t mention TPM even once, so it's not really relevant to the discussion.

Totally get the general distrust of things like Intel ME, but TPM is a separate piece of tech with different implications.

Other than that: I also think we don't need a TPM, especially in the first iterations.

BigBOSSJake, 2025-08-24:
Yes , MarcSN311 I think that you are correct, the video even if we state as truth does not talk about TPM , and yes, we do not need it in the first prototype, just some thing that might get added later.