SPIRIT-org/SPIRIT #50: Power management problems with raspberry pi family

berkerogluu, 2025-08-15:
An idle pi 5 will consume 1W to 2W, raspberry pi is not designed for mobile usage. Some people suggested that governing cpu, undervolting etc will work but sadly it is far from reality. Software, kernel optimizations will not work. Raspberry pi has no true suspend-to-ram, hibernation support.. no proper power management system. It is designed to work plugged in. I think we should consider switching to orange pi, taking action now will save time and labor in the future. Some can argue we can switch to better boards later but I think we need an unified approach from start.
Orange Pi 5 also has mainline kernel support (I don't know if it's fully competed or reliable) and android.

6mat, 2025-08-15:
I think this is a very valid issue. With moderate usage and current battery set up, it'll be l draw like 4 to 5 W. So it'll last only like 2 to 3 hours. And that's considering only the Pi5. Adding screen and everything else, it'll probably be less than an hour for moderate usage

V3lectronics, 2025-08-16:
Very valid point raised here, however changing the CM now will really slow us down with the 1st prototype. We are considering these options as potentially better alternatives:

https://github.com/V3lectronics/SPIRIT/wiki/SoC-and-SoM-alternatives-to-CM5

(not an exhaustive list, contributions welcome)

easton57, 2025-08-16:
Long term for both CM5 and anything with Rockchip to keep in mind is kernel support and development. Raspberry Pi has the raspberry pi foundation behind it and a large community including other large companies and hobbyists while, Rockchip linux development was backed by a single guy who has since backed down due to burnout and a lack of support from Rockchip and other parties who were using his work for their own products. While short term something with a Rockchip may be better, software compatibility, stability and security may become a bigger concern than finding a workaround for the raspberry pi's poor power management. The rockchip situation may have changed though, I haven't heard anything too crazy since Joshua Riek stepped away last November.

https://www.jeffgeerling.com/blog/2024/popular-rockchip-sbc-distro-limbo-after-maintainer-burns-out

easton57, 2025-08-16:
There are of course others like Armlogic and some risc v boards are popping up as well.

easton57, 2025-08-29:
I will be working on this soon. Waiting on some hardware.