SPIRIT-org/SPIRIT #52: Feature Request: Internal Charging Port

Minerik007, 2025-08-15:
# Why?
- People might want to install an additional battery or solar panel.
- This would allow sliding in a solar panel or other charging module easily.
- The only modification needed would be to reprint the back of the chassis.

V3lectronics, 2025-08-16:
1) So essentially, what you are suggesting is a secondary power input? @Minerik007 

2) Are you suggesting to place it inside the chassis?

Minerik007, 2025-08-16:
> 1. So essentially, what you are suggesting is a secondary power input? [@Minerik007](https://github.com/Minerik007)
Yes.

> 2. Are you suggesting to place it inside the chassis?
Yes. Somewhere like here:
[image]
The port will likely face upward in this case, or downward depending on how much space is available. Ideally, we want to avoid having cables from solar panel or similar power input hanging outside the phone.

closed 2025-09-01 by V3lectronics