SPIRIT-org/SPIRIT #55: Using pogo pins and small latches to easily swap between batteries.

DanR824, 2025-08-16:
You can add a smaller Battery in parallel so that no power is lost, a seamless battery swap with out interruption.
[image]

DansDesigns, 2025-08-17:
very interesting idea!
In theory you could put the phone to hibernate, have a super capacitor charge up to hold enough power for the hibernation state, disconnect the battery circuitry at the same time as the physical locking tabs are moved to their "unlocked" position and swap the batteries.. when the locking tabs are moved back into "locked" potision, the battery reconnects to take over, (recharges the super cap?) and the phone can be woken up by the power button (not sure your exact layout on power triggering)

DansDesigns, 2025-08-29:
i found a couple of replaceable batteries in the style you suggested (plastic case), alot of the ones available now have the small flex connector with the soft pouch unprotected...

heres the plastic case ones:
5200mah Samsung S4:
http://allbatteries.co.uk/mobile-phone-battery-pour-samsung-3-7v-5200mah-gml90425.html

3100mah Samsung Note3:
https://www.allbatteries.co.uk/mobile-phone-battery-pour-samsung-3-8v-3100mah-gml901518.html