SPIRIT-org/SPIRIT #58: Suggestion: Watertight USB-C port

MavLeague, 2025-08-20:
Hello, so the current usb-c connector isn't watertight and also it had an old usb standard. But We've found a better usb-c port that is watertight according to [this datasheet](https://www.hirose.com/product/document?clcode=CL0480-0522-0-00&productname=CX60-SLDA&series=CX&documenttype=Catalog&lang=en&documentid=D52488_en).

[image]

( I'd prefer the mid mount variant over the bottom mount variant, because then it is aligned with the audio jack in the chassis.) 

It also already has 3d data for download so we don't need to make an extra model for that.

[image]

Here we can find it on digikey: https://www.digikey.de/en/products/detail/hirose-electric-co-ltd/CX90MW9-24P1/22607398?s=N4IgTCBcDaIMIA0CcAGAsgdSQWjAFgAUBGEAXQF8g

[image]

**If you can find a better connector or have any ideas/ suggestions, please let us know.**

V3lectronics, 2025-08-21:
looking good I'll add it to kicad

V3lectronics, 2025-08-21:
added in KiCad!

closed 2025-08-22 by V3lectronics