SPIRIT-org/SPIRIT #60: Idea for a built-in physical keyboard

RedstoneTrail, 2025-08-22:
I'm not sure how would be the best way to do this, but having the option to add a physical keyboard to the phone would be a nice option in my opinion.

I had 2 ideas of how to do this though because each has its own advantages and disadvantages.

||Blackberry-style|iPad-style|
|-|-|-|
|Description|The full-height screen is replaced with a half-height screen with a small keyboard below it taking up the rest of the length of the phone|The keyboard is folded behind the phone when out of use / can be removed and re-attached (maybe magnetically or using the resistance from a usb port|
|Benefits|Can be used in your hands with your thumbs|Still keeps a full size screen and there is a bigger keyboard (ideal for typing against a table)|
|Disadvantages|Smaller screen|Makes the phone thicker, needs to be on a table to use properly|

maybe an internal usb port could be exposed for this kind of thing or something like that

let me know what you think!

Secret-chest, 2025-09-03:
I'd suggest something like the formerly available PinePhone keyboard addon

DansDesigns, 2025-09-04:
theres also Clicks by Mr. Mobile on youtube

Secret-chest, 2025-09-06:
I suggest pogo pins implementing USB but in a different shape.

RedstoneTrail, 2025-09-06:
Thats not a bad idea, but i think it was found that that kind of solution could be too expensive (there was an issue discussing an expansion connector thing using a magnetic connector but that was a bit too expensive to buy the connectors). I think a proper usb 2 port might be a better solution, maybe it could be embedded in the back with a cutout/lowered section in the case for access from the back. other than if there are price issues, this does sound like a good solution

Secret-chest, 2025-09-06:
That could also work, imagine the keyboard having a hinge with an USB plug sticking out of it

gitmanbloggus, 2025-09-26:
or... some thunderbolt-like port to expose pcie