SPIRIT-org/SPIRIT #66: Snap on / Slide out Keyboard

FXDSSTM, 2025-09-01:
### RND Keyboard and how to implement it with SPIRIT phone

Potential options for connectivity:
- USB
- I2C

Keyboard options:
- Slide out mechanism integrated into chassis
- Snap on attachment (pogo pins or plug in)

Options for microcontrollers:
- ATMega32U4
- ATMega8A

Koakno, 2025-09-01:
If it's snap on, wouldn't you want to use usb for hot swapping?

(Or am I mistaken?)

Secret-chest, 2025-09-03:
What does RND mean? Also I am very interested: where can such a sized keyboard be sourced? BlackBerry keyboards are too small and not sustainable to use.

FXDSSTM, 2025-09-04:
> What does RND mean? Also I am very interested: where can such a sized keyboard be sourced? BlackBerry keyboards are too small and not sustainable to use.

RND - Research & Develop. The idea is to create a custom keyboard from scratch that can either be implemented into a chassis design for those that want to have both screen and keyboard, or can be an "add on" for those that would like to take it with them without taking up any of the phone's space. Planned keyboard size would be more along the lines of the phone's length, but would also be able to go according to width if people want to slide up instead of to the side.

closed 2025-09-17 by V3lectronics