SPIRIT-org/SPIRIT #69: How do you design the I/O boards?

Secret-chest, 2025-09-03:
I am quite interested as I want to learn more and possibly contribute/design an alternative project to this (whilst still hopefully collaborating). I have never done PCB design before, but what are good resources for making custom CM I/O boards?

Secret-chest, 2025-09-04:
Not the 3D, simply the rules that you have to follow when drawing the connections

DansDesigns, 2025-09-04:
thats a great idea, I too am working on an CM4 compatible adaptor for Pi Zero style SBCs for my OSM-Phone

V3lectronics, 2025-09-09:
> but what are good resources for making custom CM I/O boards?

I'd recommend these 2 things:
- https://www.youtube.com/watch?v=ge6gYIENo8Q
- get the original RPi CM5 IO board EDA files and modify it to your need.

V3lectronics, 2025-09-17:
closing as inactive

closed 2025-09-17 by V3lectronics