SPIRIT-org/SPIRIT #76: Check EG25 symbol

V3lectronics, 2025-09-19:
The task is to check if the EG25 symbol in the GPS GSM schematic (value: `EG25GGC-128-SGNS`) is correct for the 256 variant of the EG25 as well.

Product page:
https://www.digikey.pl/en/products/detail/quectel/EG25GGB-256-SGNS/21272668

If yes, please rename the symbol and footprints so that there is no more confusion between the variants.

BigBOSSJake, 2025-09-19:
Take a look(this is a good website too 👀).[EG25GGB-256](https://www.snapeda.com/parts/EG25GGB-256-SGNS/Quectel/view-part/)  ,[EG25GGC-128](https://www.snapeda.com/parts/EG25GGC-128-SGNS/Quectel/view-part/)

V3lectronics, 2025-09-22:
Ok I checked all 3 sections of the symbol, I see both are the same on the website and the same as the one we have in KiCad except:
- current KiCad has multiple vbat pins
- reserved pins
- separate gnds (idk if that even matters in terms of symbols)

So I would stick with the one we have currently!

closed 2025-10-01 by V3lectronics